ModbusScope/ModbusScope #92: Maximum between cursors isn't calculated correctly

pluyckx, 2017-04-10:
When calculating the maximum between cursors, it is always 1 for small signals. I checked the code and you should change the following on line 283 to something like:
double max = -INFINITY;

Maybe you should also change line 269 to calculate the minimum to something like:
double min = INFINITY;

Of course you need to include a header for these defines. In C++ there is something like <limits>. The examples in the code above is for C, for C++ there is an implementation with templates if I am correct.

pluyckx, 2017-04-10:
Have a look at the following site to get the correct min/max values in C++:
http://en.cppreference.com/w/cpp/types/numeric_limits

jgeudens, 2017-08-02:
Fixed in pull request (8ee50dc733ffa35bbbbb3bccb1fad94ff6e09df3)

closed 2017-08-02 by jgeudens