ModbusScope/ModbusScope #147: Make it possible to show wall clock time in x-scale

ben-edna, 2021-01-29:
When doing a test and trying to cross-reference a issue with another log, it would be easy if the wall clock time is shown.

jgeudens, 2021-01-29:
Do you mean absolute time?

https://modbusscope.readthedocs.io/en/latest/pages/configuration.html#configure-log-settings

spoorcc, 2021-01-29:
Yes indeed, but then in the graph

jgeudens, 2021-01-29:
I'm not sure what you mean. When absolute times is enabled, the log shows absolute time and date, but also the graph.

[image]

spoorcc, 2021-01-29:
Ouch, didn't knew that, please consider this closed. I can make a certain customer happy 😁

closed 2021-01-29 by ben-edna