ModbusScope/ModbusScope #150: Import MBC: Enable multiple registers

jgeudens, 2021-02-26:
Wouldn't it be nice to select multiple registers from an MBC file and enable all of them with a single action?

ben-edna, 2021-03-06:
I don't completely understand, it is possible to select them during import isn't it?

jgeudens, 2021-03-07:
Yes, that is true. But this specific use case present itself when many (more than 10) subsequent registers need to be selected for importing. At the moment, the user needs to click the checkboxes separately. If it would be possible to easily select all these register with the mouse and then use a button or something to check the import checkboxes of these selected registers.

The issue was reported by an user.

spoorcc, 2021-03-07:
Ah, you mean a "select all" button?

jgeudens, 2021-03-07:
Yes, something similar. Not select all registers that are visible in the list. But more, check all selected (blue) rows.

closed 2022-10-18 by jgeudens