ModbusScope/ModbusScope #151: Enconding issue in MBCFileImporter

ben-edna, 2021-03-04:
I have a mbc file with the following entry:

```xml
<?xml version="1.0" encoding="utf-8"?>
...
    <var><reg>40001</reg><text>Estimated System Flow [m³/h]</text><decimals>3</decimals><type>uint32</type><rw>r</rw><bcol>255,225,255</bcol></var>
```

In ModbusControl this is shown correct:
[image]

But in ModbusScope, I think something goes wrong with the encoding:

[image]

My guess is that since the mbc is read as plain text file in [`ImportMbcDialog::updateMbcRegisters()`]( https://github.com/jgeudens/ModbusScope/blob/8b05404fa868fa8dac9e56b4ce51efb165249aa1/src/dialogs/importmbcdialog.cpp#L138-L139) and the content is passed as a string no encoding detection is done by QDomDocument.

jgeudens, 2021-03-05:
Thank you the report. I will check it out.

jgeudens, 2021-03-05:
Hmm, I can't seem to reproduce the issue on Windows (or Ubuntu). Are you using the latest release (v3.1.0)?

[image]

ben-edna, 2021-03-06:
On windows with this version: 
[image]

Attached a minimal reproducing file with the error:
[repro.zip](https://github.com/jgeudens/ModbusScope/files/6095071/repro.zip)

(Sidenote: I also noticed when the `<rw>` tag is missing only an "invalid" message is given, not that it it missing a field.)

ben-edna, 2021-03-06:
I also upgraded to v3.1.0 and issue is still present

spoorcc, 2023-02-17:
I no longer see this in v3.7.0:

[image]

jgeudens, 2023-02-18:
Thanks for testing again.

closed 2023-02-18 by jgeudens