ModbusScope/ModbusScope #181: Feature Request: Allow full 16-bit addressing of Holding Registers

SweJake, 2022-01-18:
Hi,

I was very happy when i stumbled upon this project!
Just what I was looking for to read/log my new HeatPump.
But...
The interesting Modbus Holding Registers of my manufacturer all start above standard.
Please see:
https://ctc.se/file-download/a9fcdab53f94cdb3ae70beb27e9372b7.pdf
(Ex. "Outdoor temp." = Holding Reg# 62147)
[image]
(I've tried to tweak the settingsfile, *.mbs, to enter higher value but can see in WireShark that it doesn't work)
I would so much like to have this working with my HeatPump.
So..

**Feature Request:**
Add a checkbox stating: "Allow out of range Holding Registers" or similar ?

jgeudens, 2022-01-19:
I'm glad that ModbusScope might be useful for you.

I'm currently working on making ModbusScope more generic, separating Modbus registers from a data graph using expressions and such. Your feature request will be useful to support more devices in the field which don't adhere very strictly to the standard.

I think it should be doable. I hope to take some time this week for an initial implementation.

jgeudens, 2022-01-19:
I have added a test to read a large register address. During the refactoring I mentioned, I already removed some obsolete checks on the register address. Apparently, I already support registers 40001 to 65353 in the current beta version. You can download this beta from my CI build artifacts (https://ci.appveyor.com/project/jgeudens/modbusscope/builds/42262445/artifacts).

You can use following expression to read register 61500 with the new syntax.

[image]

Of course, this isn't the complete range since 40001 is just an offset for holding registers that isn't used in the protocol itself so the full range is 40001 to 105536. but allowing this would require some more changes.

I hope you can use ModbusScope with your device with this build. I would appreciate some feedback whether it worked or not.

Good night.

jgeudens, 2022-01-21:
Support for the full 16-bit range is merged through #185

spoorcc, 2022-01-24:
Hi @SweJake,

If I'm not mistaken the 40001 offset is always subtracted in a read holding register message (function code 3).  See also http://www.simplymodbus.ca/FC03.htm.

The packets contents seems correct to me.

SweJake, 2022-01-25:
Oh, I think we misunderstood each other, and that's mostly my fault for being unclear.

In the document:
https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf
at chapter: "4.3 MODBUS Data model"
They are talking about the "normal" 4-block device, which I believe this program supports excellently!
...but later on they also describe the 1-block device. (mine)

So, if i want to read ANY register i have to send a read request ,0x03 "Read holding reg." preferably, followed by the full 16-bit reg-address:
[image]

And i will get the "OutDoor Temp." (and divide it by 10 --> 1.5℃)
[image]

See the document of my manufacturer in the original request.

So maybe the title of the request would have been: "Please add support for 1-block ModBus devices"?

jgeudens, 2022-01-25:
Today I learned that modbus has a 1-block model.

As long as the data is available through the read holding register function code (FC3). You should be able to use ModbusScope. With this [PR](https://github.com/jgeudens/ModbusScope/pull/185), I have added full support for the 16-bit adressing. You can download the latest build [from the CI](https://ci.appveyor.com/project/jgeudens/modbusscope/builds/42292945/artifacts).

I must admit I don't really understand the register addressing of your device, so I can't help you there. But you should be able to use ModbusScope to read all data. ModbusScope will substract the 40001 offset automatically from the address. But you can work around this by adding 40001 to the requested address.

* `${40001}` will give you raw address 0 in the Wireshark packet.
* `${102148}` will give you raw address 62147 in the packet.

I hope this information is useful for you.

SweJake, 2022-01-27:
Somewhat awkward to enter the reg-addresses, but i can live with that.
I'm completely satisfied!
This supports my setup of a: CTC-EcoAir408 and CTC-EcoLogic S.
I will give it a thoroughly "Burn-In Test"
[image]

Big Thanks !

closed 2022-01-31 by jgeudens