ModbusScope/ModbusScope #250: Reading 32 bit floating point data from Modbus registers

GJMorrison, 2022-11-23:
The device I am trying to graph contains a mixture of 16 bit signed and unsigned data, 8-bit unsigned data and 32 bit floating point data.
I cannot find any way to read the 32 bit floating point values in from the modbus device in the Register Settings.

jgeudens, 2022-11-24:
Duplicate of https://github.com/jgeudens/ModbusScope/issues/226

At the moment, ModbusScope doesn't support floating numbers. But I think it would be possible to add a float32 type.

Would you or @whyily be interested in providing test cases for this?

I would expect that following edge cases would be present:
* maximum positive value
* minimum positive value
* maximum negative value
* minimum negative value
* -1
* 0
* ...

You can also add other sensible values

If you want, you can provide these in following form:
| First register  | Second register | Float |
| ------------- | ------------- | ------------- |

spoorcc, 2022-11-24:
There is a nice summary on Wikipedia with some notable cases with the hex representation next to it. 
 https://en.wikipedia.org/wiki/Single-precision_floating-point_format#Notable_single-precision_cases

jgeudens, 2022-11-25:
Also interesting: https://en.cppreference.com/w/cpp/numeric/math/fpclassify

```
switch(std::fpclassify(x)) {
    case FP_INFINITE:  return 0.0f;
    case FP_NAN:       return 0.0f;
    case FP_NORMAL:    return x;
    case FP_SUBNORMAL: return x;
    case FP_ZERO:      return 0.0f;
    default:           return x;
    }
```

jgeudens, 2023-01-03:
@spoorcc @GJMorrison 
I have added the `f32b` type to read 32-bit floating point registers.

If you want, you can download an development version from the artifacts: https://github.com/jgeudens/ModbusScope/actions/runs/3832038958

GJMorrison, 2023-01-04:
@jgeudens 
I have tried your floating point build and so far it seems to be working as expected with the floating point data.
Great job!
Thanks.

spoorcc, 2023-01-04:
@jgeudens 

Great job!  Works as expected, only the import functionality needs some work. But I can overcome that for now 👏

jgeudens, 2023-02-17:
Remaining items are implemented and released in https://github.com/jgeudens/ModbusScope/releases/tag/3.7.0

closed 2023-02-17 by jgeudens