ModbusScope/ModbusScope #277: Read other Modbus registers

FranVojvoda, 2023-05-18:
Hi! I'm totally new to modbus systems, but I'm trying to log data from a controller with your app. I find that not all the data I need is inside the Holding Register, I want to register data inside Input Status (0/1), I have tried to point directly to that address without success.

[image]

Congratulations on your application, it is very complete and possibly the most stable I have found.

Thank you

jgeudens, 2023-05-18:
Thank you for the kind words. I do my best :wink: 

At the moment, only holding registers are supported by ModbusScope, so I'm afraid it isn't possible to read input registers or coils.

FranVojvoda, 2023-05-22:
Hi! Thanks for your answer!

Maybe for a future version it would be interesting to add it... As well as the possibility of writing records.

Or that's what I would like. 🤣 

Also, thank you very much for your work!

jgeudens, 2023-11-03:
If you're still interested. I just merged support for other object types (coil, discrete input and input register). The type will be determined based on the address.

Coil/Register Numbers | Table Name
-- | --
1-9999 | Discrete Output Coils
10001-19999 | Discrete Input Contacts
30001-39999 | Input Registers
40001- xxx | Holding Registers

alpytron, 2023-12-02:
Hi @jgeudens,
May I ask you to create a release containing the support of reading those registers you mentioned above? That would really be appreciated.
Thanks in advance.

alpytron, 2023-12-05:
This works perfectly.
Thank you so much.