ModbusScope/ModbusScope #292: documentation doesn't describe how to create mbc (Modbus Configuration) file

dalers, 2023-11-22:
Minor shortcoming, the manual explains how to import an mbc file, but doesn't explain how to create one. Do I just change the extension of an mbs project file? Do I use the top part of a CSV output file?

Cheers (and really loving ModbusScope after having it recommended),
Dale

jgeudens, 2023-11-24:
Always glad to hear from a happy user.

A mbc file is a project file from another application called ModbusControl. This is a proprietary application that isn't available for the general public. It is created and used internally at my work. But you aren't missing any functionality in ModbusScope. A mbc file just contains definitions of modbus registers that can be imported in ModbusScope. These registers are then saved to the normal mbs file.

I will improve the documentation to avoid further confusion. Maybe it is better that the import mbc option is a bit more hidden in a future version of ModbusScope.

dalers, 2023-11-24:
Thanks @jgeudens for responding.

> Maybe it is better that the import mbc option is a bit more hidden in a future version of ModbusScope.

That would be disappointing :-(  Not having to not manually configure ModbusScope each time would be extremely convenient and really add value.

Would it be possible to specify the mbc file format sufficiently in order for a user to manually create one? I skimmed importexport.cpp to see if the file format might be obvious, but unfortunately I'm more of a user than a coder and quickly got lost :-( 

Alternatively, could I suggest adding an mbc export function to ModbusScope to create just enough of an mbc file to define the registers? 

Cheers

jgeudens, 2023-11-24:
I think you might misunderstood my comment. The mbc file is a project file for another application.

ModbusScope uses its own mbs project file. When you configure ModbusScope with the relevant registers, you can save the configuration as mbs file. This mbs file can later be reopened by ModbusScope to get the same configuration.

[Storing and reusing configuration](https://modbusscope.readthedocs.io/en/stable/pages/importing_and_exporting.html#storing-and-reusing-configuration)

The configuration of registers, as described in previous sections, can be saved and loaded in a project file. These settings can be saved in a .mbs file by going to File > Save Project As… and loaded by going to File > Open Project… or by dragging a .mbs file into the application.

dalers, 2023-11-25:
Ah, yes, understood thanks! My apologies for responding prematurely.

closed 2023-11-25 by dalers

jgeudens, 2023-11-27:
Like I said, glad that I could help.