ModbusScope/ModbusScope #304: Axis loses focus when `ctrl + left clicking` to set first marker

pluyckx, 2024-01-11:
**Description:**
When you have selected the x-axis (or y-axis) to only zoom the x-axis (or y-axis), the selection is lost when `ctrl + left clicking` to set the first marker. This is not the case when setting the second marker using the right mouse button.

**Expected behavior:**
Axis selection is not lost when `ctrl + clicking` in the graph to set the marker. You have selected the x-axis or y-axis with a reason.
Clicking without `ctrl` removes the axis selection.

**Workaround:**
I have found a workaround for this: `ctrl + click` to set the marker, but before releasing the mouse button, move the mouse a bit. The marker will stay at the position where you pressed the mouse button down, but the axis selection is not lost when releasing the mouse because of the mouse movement.

jgeudens, 2024-01-14:
I have implemented a fix in #305. Can you verify whether the fix (https://github.com/ModbusScope/ModbusScope/actions/runs/7518521514) is working for you?

Thank for mentioning the workaround, it gave very useful information to find the issue.

jgeudens, 2024-02-03:
Merged with #305

closed 2024-02-03 by jgeudens