ModbusScope/ModbusScope #360: Feature Request: Scalable Y-Axis Per Register

sabrina-dm, 2025-05-15:
Hello,

We appreciate the work that has gone into this project—it has been incredibly useful for visualizing Modbus data.

Currently, when working with multiple registers, the varying value ranges make some datasets difficult to interpret. 
To improve readability, we’d like to request a scalable Y-axis per register, allowing each dataset to be displayed appropriately without being overshadowed by others.

Would this be feasible within the current setup? We're open to discussing implementation approaches and contributing if needed.

Looking forward to your feedback.

jgeudens, 2025-05-16:
Hi,

I'm glad to hear that ModbusScope has been useful for you.

In the beginning, there was only one Y-axis and I encountered the same issues. My workaround was to scale (multiply, divide, offset) the values using the expressions to make sure they were in the same range. With the addition of the secondary Y-axis, it was a great improvement. And to be honest, it is enough for most of my use cases.

But if you're willing to put in some work, I think it is possible to add more Y-axis. There are some things to consider:

- Visually:
  - Extra y-axis can be added to the right side. Similar to the screenshot in the QCustomPlot examples (https://www.qcustomplot.com/index.php/tutorials/specialcases/axistags)
- Configuration:
  - The y-axis dock already has a tab per Y-axis, so these can dynamically be added according to the number of Y-axis.
  - But the widget is designed using a UI file, this should probably be converted to a programmatic approach.
- Project file:
    - The xml of the MBS file also supports more Y-axis. The relevant tags have a index attribute.

Just let me know, if you're still interested.

jgeudens, 2026-03-28:
Hi, this topic is stale for some time now. I'm also working on an complete overhaul of the application, so I 'm not sure if it is still valuable to keep this issue open.

I'm also reconsidering the open source model with contributions. The application will remain open source, but with the advent of all AI assistents. I'm thinking about closing the PR section for external contributions. I'm not interested in maintaining or reviewing lots of AI generated code (except for the code I generated myself 😉).

closed 2026-03-28 by jgeudens